lvgl/lv_lib_gif #5: CANT FIND DEFINE

chliii, 2020-09-15:
[image]

chliii, 2020-09-15:
[image]
run on the CLION for ESP32

kisvegabor, 2020-09-16:
As stated in the README you need to use the `feat/new-fs-api` branch of LVGL with the GIF decoder.

chliii, 2020-09-17:
Thanks your answer.

But,when I run the program, the screen show "No data".

    lv_obj_t * img = lv_gif_create_from_file(lv_scr_act(),"F:/ESP32/esp-gif/src/ship.gif");

Maybe I set a wrong path?

chliii, 2020-09-17:
[image]
[image]

And,did I set the path is true ?
lv_obj_t *img = lv_gif_create_from_file(lv_scr_act(), "p/ship.gif");
    lv_obj_align(img, NULL, LV_ALIGN_CENTER, 0, 0);

chliii, 2020-09-19:
the function "ftruncate" cant compile on the ESP32.

#if defined(__CYGWIN__) || defined(__rtems__) || defined(__aarch64__) || defined (__arm__) || defined(__sh__) || defined(__SPU__)
#if !defined(__INSIDE_CYGWIN__)
int     _EXFUN(ftruncate, (int __fd, off_t __length));
int     _EXFUN(truncate, (const char *, off_t __length));
#endif
#endif

kisvegabor, 2020-09-21:
Hm, if `ftruncate` is a standard C function however it's important to support ESP32. If you remove this function the rest is working?

chliii, 2020-09-24:
Hi,
if I remove this function the rest is working,but,the screen only show _No data_.

chliii, 2020-09-24:
My WARNING and the screen
[image]
[image]

kisvegabor, 2020-09-30:
How do you set the path of the gif?

Please enable the [logging of LVGL](https://docs.lvgl.io/latest/en/html/porting/log.html) to see what's going on.

closed 2020-10-22 by chliii