lvgl/lv_lib_gif #7: If I just want to use the c_array to display the GIF

KiraVerSace, 2020-10-16:
If I just want to use the c_array to display the GIF, would you like to simplify the  code, and remove the lv_fs_if, it's difficult for me to do it.

because I add the lv_fs_if and lv_lib_gif library and enable 
```
/* 1: Enable file system (might be required for images */
#define LV_USE_FILESYSTEM       1	// FIXME:0->1
```

and add

```
/*File system interface*/
#define LV_USE_FS_IF	1
#if LV_USE_FS_IF
#  define LV_FS_IF_FATFS    '\0'
#  define LV_FS_IF_PC       '\0'
#endif  /*LV_USE_FS_IF*/
```

there also many errors in the compile. Sorry, Could you help me, and I am from China, and do not good at English.

Thank you for your contribution!

[image]

kisvegabor, 2020-10-19:
I can update the gif library to work well `LV_USE_FILESYSTEM 0` (without `lv_fs_if`).

Would it work for you?

KiraVerSace, 2020-10-19:
> I can update the gif library to work well `LV_USE_FILESYSTEM 0` (without `lv_fs_if`).
> 
> Would it work for you?

no if you dont use the lv_fs_if it will be more errors, I include the two folders.

Now in my project I use 20 frams/second to display the animations, haha!

kisvegabor, 2020-10-19:
I meant to make the gif library work with `LV_USE_FILESYSTEM 0` without `lv_fs_if` or any other FS related stuff.

KiraVerSace, 2020-10-19:
yes I have try it set LV_USE_FILESYSTEM 0, but errors happens too

kisvegabor, 2020-10-19:
I've pushed a fix, please try with the latest version,

KiraVerSace, 2020-10-19:
> I've pushed a fix, please try with the latest version,

thank you for your contribution, but I doubt if I use the gif library or use too many images like 20fps, which will be better and which use less flash size?

kisvegabor, 2020-10-19:
gif is compressed fo needs less flash but slower becasuse needs decompression.
gif has a 8 bit palette so it can have only 256 colors per frame.

Finally are you happy with 20FPS or is it too slow for you?

chliii, 2020-10-22:
朋友 可以加个联系方式吗

chliii, 2020-10-22:
@KiraVerSace 朋友 可以加个联系方式吗

closed 2020-11-26 by stale[bot]