lvgl/lv_lib_gif #9: Does my code is wrong?

chliii, 2020-10-22:
#include <Arduino.h>
#include <TFT_eSPI.h>
#include <lvgl.h>
#include "lv_gif.h"

TFT_eSPI tft = TFT_eSPI();
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];
LV_IMG_DECLARE(test);
LV_IMG_DECLARE(stun);

void my_flush_cb(lv_disp_drv_t *disp, const lv_area_t *area,
                 lv_color_t *color_p) {
    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    tft.startWrite();
    tft.setAddrWindow(area->x1, area->y1, w, h);
    tft.pushColors(&color_p->full, w * h, true);
    tft.endWrite();
    lv_disp_flush_ready(disp);
}

void setup() {
// write your initialization code here
    Serial.begin(115200);

    lv_init();

    tft.begin();
    tft.setRotation(1);
    lv_disp_buf_init(&disp_buf,buf,NULL,LV_HOR_RES_MAX * 10);

    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = 320;
    disp_drv.ver_res = 240;
    disp_drv.flush_cb = my_flush_cb;
    disp_drv.buffer = &disp_buf;
    lv_disp_drv_register(&disp_drv);

    lv_obj_t * img1 = lv_gif_create_from_data(lv_scr_act(), &test);
    lv_obj_align(img1,NULL,LV_ALIGN_CENTER,0,0);

}

void loop() {
// write your code here

    lv_task_handler();
    delay(5);
}

chliii, 2020-10-22:
My screen still can't display the GIF.

:(

kisvegabor, 2020-10-26:
Have you enabled logging as described here? 
https://docs.lvgl.io/latest/en/html/get-started/arduino.html

chliii, 2020-10-27:
I try to printf my log.
`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Trace: lv_init started 	(lv_obj.c #135 lv_init())
Info: lv_init ready 	(lv_obj.c #186 lv_init())
Trace: Screen create started 	(lv_obj.c #228 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Trace: Screen create started 	(lv_obj.c #228 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Trace: Screen create started 	(lv_obj.c #228 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Trace: image create started 	(lv_img.c #62 lv_img_create())
Trace: Object create started 	(lv_obj.c #260 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Info: image created 	(lv_img.c #127 lv_img_create())
invalid signature
Info: lv_img_set_src:  `LV_IMG_SRC_VARIABLE` type found 	(lv_img.c #180 lv_img_set_src())
Trace: lv_task_handler started 	(lv_task.c #74 lv_task_handler())
Trace: lv_refr_task: started 	(lv_refr.c #177 _lv_disp_refr_task())
Trace: lv_refr_task: ready 	(lv_refr.c #310 _lv_disp_refr_task())
Trace: lv_task_handler ready 	(lv_task.c #192 lv_task_handler())
Trace: lv_task_handler started 	(lv_task.c #74 lv_task_handler())
Trace: lv_task_handler ready 	(lv_task.c #192 lv_task_handler())
Trace: lv_task_handler started 	(lv_task.c #74 lv_task_handler())
Trace: lv_task_handler ready 	(lv_task.c #192 lv_task_handler())
`

chliii, 2020-10-27:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Trace: lv_init started 	(lv_obj.c #135 lv_init())
Info: lv_init ready 	(lv_obj.c #186 lv_init())
Trace: Screen create started 	(lv_obj.c #228 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Trace: Screen create started 	(lv_obj.c #228 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Trace: Screen create started 	(lv_obj.c #228 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Trace: image create started 	(lv_img.c #62 lv_img_create())
Trace: Object create started 	(lv_obj.c #260 lv_obj_create())
Info: Object create ready 	(lv_obj.c #417 lv_obj_create())
Info: image created 	(lv_img.c #127 lv_img_create())
invalid signature
Info: lv_img_set_src:  `LV_IMG_SRC_VARIABLE` type found 	(lv_img.c #180 lv_img_set_src())
Trace: lv_task_handler started 	(lv_task.c #74 lv_task_handler())
Trace: lv_refr_task: started 	(lv_refr.c #177 _lv_disp_refr_task())
Trace: lv_refr_task: ready 	(lv_refr.c #310 _lv_disp_refr_task())
Trace: lv_task_handler ready 	(lv_task.c #192 lv_task_handler())
Trace: lv_task_handler started 	(lv_task.c #74 lv_task_handler())
Trace: lv_task_handler ready 	(lv_task.c #192 lv_task_handler())
Trace: lv_task_handler started 	(lv_task.c #74 lv_task_handler())
Trace: lv_task_handler ready 	(lv_task.c #192 lv_task_handler())

chliii, 2020-10-27:
I try to printf my log

kisvegabor, 2020-10-27:
How many RAM does your MCU have?
Could you try with a really small gif? (e.g. 20x20 px)

closed 2020-12-04 by stale[bot]