lvgl/lv_lib_gif #12: Cant seem to get it to compile

elgerg, 2021-01-03:
I have taken the latest feature version of the lv_fs and enabled it:

/*File system interface*/
#define LV_USE_FS_IF	1
#if LV_USE_FS_IF
#  define LV_FS_IF_FATFS    '\0'
#  define LV_FS_IF_PC       'S'
#  define LV_FS_IF_POSIX    '\0'
typedef void * lv_fs_drv_user_data_t;
#endif  /*LV_USE_FS_IF*/

However when trying to compile it i get:

../lv_lib_gif/gifdec.c:504:19: error: too few arguments to function ‘lv_fs_open’
         gif->fd = lv_fs_open(path, LV_FS_MODE_RD);

If i set the original file system option to 0 in the conf:

#define LV_USE_FILESYSTEM       0 

I got a whole load more errors. 

What am i doing wrong?

kisvegabor, 2021-01-04:
Do you use the `new-api` branch?

elgerg, 2021-01-04:
I believe so. I downloaded this:

[image]

Or is that wrong?

Thanks
Alex

kisvegabor, 2021-01-04:
Take a look at the README of https://github.com/lvgl/lv_lib_gif

> Compatible with LVGL's feat/new-fs-api branch. If you use the lv_fs_if repository its new-api branch should be used.

elgerg, 2021-01-04:
Ok. I think i had added that separately where I shouldnt have?

Now it compiles but show an image with No Data in it. The console logs:

Warn: Image draw: src is NULL 	(lv_draw_img.c #77 lv_draw_img())
Warn: lv_img_design: image source type is unknown 	(lv_img.c #728 lv_img_design())

Am I missing something with regards to drivers or initializing them?

I have the following in the config:

/*File system interface*/
#define LV_USE_FS_IF	1
#if LV_USE_FS_IF
#  define LV_FS_IF_FATFS    '\0'
#  define LV_FS_IF_PC       '\0'
#  define LV_FS_IF_POSIX    'L'
#endif  /*LV_USE_FS_IF*/

Ive tried setting IF_PC and IF_POSIX to L with the following in my code:

````c

    /*Temp show image*/
    lv_obj_t * img1 = lv_gif_create_from_file(parent, "L/home/alex/Workspace/LV_HACP/App/Gifs/images.gif");

    lv_obj_align(img1, NULL, LV_ALIGN_CENTER, 600, -20);

````

elgerg, 2021-01-04:
Oh, ive just noticed something. Let me change that and try again.

elgerg, 2021-01-04:
No, I thought my path was wrong but it's not.

Any ideas?

elgerg, 2021-01-04:
It looks like this might be the problem 

````c
static bool f_gif_open(gd_GIF * gif, const void * path, bool is_file)
{
    gif->f_rw_p = 0;
    gif->data = NULL;
#if LV_USE_FILESYSTEM
    gif->fd = NULL;
#endif

    if(is_file) {
#if LV_USE_FILESYSTEM
        gif->fd = lv_fs_open(path, LV_FS_MODE_RD);
        if(gif->fd == NULL) return false;
        else return true;
#else
        return false;
#endif
    } else
    {
        gif->data = path;
        return true;
    }
}
````

If you turn off the LV_USE_FILESYSTEM null is returned. If you enable it then you get compilation errors with too few args.

Any ideas? Do I have the wrong gifdec.c?

Thanks

elgerg, 2021-01-04:
Finally figured out what you mean by:

Compatible with LVGL's feat/new-fs-api branch. If you use the lv_fs_if repository its new-api branch should be used.

Its a branch of LVGL not a branch of fs_api.

elgerg, 2021-01-04:
So. I have downloaded the new LVGL branch and that compiles. It still wont load the image.

Am i missing something with regards to drivers for linux? I have tried both PC and posix setting the letter to 'L' and used the following:

lv_obj_t * img1 = lv_gif_create_from_file(parent, "L:/home/alex/Workspace/pc_simulator_sdl_eclipse/App/Gifs/images.gif");

But no data.

It seems to fail somewhere around here:

````c
lv_fs_drv_t * lv_fs_get_drv(char letter)
{
    lv_fs_drv_t * drv;

    _LV_LL_READ(LV_GC_ROOT(_lv_drv_ll), drv) {
        if(drv->letter == letter) {
            return drv;
        }
    }

    return NULL;
}
````

Thanks for the help

kisvegabor, 2021-01-05:
Are you sure that the file system is working?  E.g. can you open a file with `lv_fs_open`?

kisvegabor, 2021-01-09:
Both the PC and POSIX versions should work on Linux. 
But first, be sure that you can open and read files without LVGL.

elgerg, 2021-01-12:
Something like:

```
        strcat(path,rom);
        strcat(path,"/w1_slave");

        // Open the file in the path.
        if((fd = open(path,O_RDONLY)) < 0)
        {
            printf("open error\n");
            return -85.00;
        }
        // Read the file
        if(read(fd,buf,sizeof(buf)) < 0)
        {
            printf("read error\n");
            return -85.00;
        }
        // Returns the first index of 't'.
        _temp = strchr(buf,'t');
        // Read the string following "t=".
        sscanf(_temp,"t=%s",_temp);
        // atof: changes string to float.
        value = atof(_temp)/1000;
        printf("Temp Retrieved: %.2f... ", (double)value);
        close(fd);
```

The above works and reads a temp from a one wire ds18b20 temp sensor.

kisvegabor, 2021-01-12:
> The above works and reads a temp from a one wire ds18b20 temp sensor.

Wasn't we talking about gifs? How did the DS18B20 come into the picture?

Please try to read gif file. The first 3 characters should be `"GIF"`.

elgerg, 2021-01-12:
Sorry, i thought you were asking if i could read the file system.

Yes gif reading works:

[image]

kisvegabor, 2021-01-13:
Great!

In `lv_fs_open` you should use `L/App/Gifs/images.gif` because LVGL assumes relative path from the working directory.

elgerg, 2021-01-13:
Ok. I must be missing something in the initialization of the file system.

[image]

[image]

[image]

[image]

[image]

I have the following include:

```
#include "lv_lib_gif/lv_gif.h"
```

Is there anything i have not set or set incorrectly?

kisvegabor, 2021-01-13:
Have you called `  lv_fs_if_init();`?

elgerg, 2021-01-13:
I did try but it says its undefined.

[image]

[image]

Am I missing an include?

kisvegabor, 2021-01-14:
Do you see `lv_fs_if_init()` [here](https://github.com/lvgl/lv_fs_if/blob/new-api/lv_fs_if.c#L52) in your code?
If you add an `#error "built"` to `lv_fs_if.c` do you see the error on build?

elgerg, 2021-01-14:
No i didnt copy the new fs source into lvgl. I didnt realise that was a requirement.

I get the image displayed now but have a separate issue with the alignment and casting.

I'll do some digging and raise another ticket if there are other problems.

Thanks for the great library and all of the help.
Alex

closed 2021-01-14 by elgerg

kisvegabor, 2021-01-14:
Glad to hear that it finally works :slightly_smiling_face: