lvgl/lv_lib_gif #13: Delete gif crash on LV_EVENT_LEAVE

FlyLu, 2021-01-09:
When there is no next frame of data, the task will send a LV_EVENT_LEAVE event
I deleted the gif object on this event,  my program is crashed.

This is original
```C
    if (has_next == 0) {
        lv_event_send(img, LV_EVENT_LEAVE, NULL);
    }
    lv_obj_invalidate(img);
    counter = ext->gif->gce.delay;

```
I modified this
```C
    if (has_next == 0) {
        lv_event_send(img, LV_EVENT_LEAVE, NULL);
    } else {
        lv_obj_invalidate(img);
        counter = ext->gif->gce.delay;
    }
```
test code
```C

  void lv_gif_event_cb(struct _lv_obj_t * obj, lv_event_t event)
  {
    if (LV_EVENT_LEAVE != event)  {
      return;
    }
  
    lv_obj_del(obj);
  }

  lv_obj_t *gif = lv_gif_create_from_file(bg, "z:test.gif");
  lv_obj_set_event_cb(gif, lv_gif_event_cb);
```
Because there is no way to know the gif is finished, except for LV_EVENT_LEAVE event

kisvegabor, 2021-01-11:
Thanks for the report, should be fixed here: 1967de3

FlyLu, 2021-01-11:
Wish LVGL do better

closed 2021-01-11 by FlyLu