lvgl/lv_lib_gif #15: Transparent background is not available

crazytyt, 2021-01-13:
I tested it with a GIF animation on NXP RT6010 board and simulator, the background is black.
The actual background color is transparent. 

When I hard code the gif->bgindex = 0xff, the background could be transparency.
It confused me a lot.  could you please verify it at your side?

-Andy

kisvegabor, 2021-01-14:
Please attach the gif you have tested with.

crazytyt, 2021-01-22:
[image]

crazytyt, 2021-01-22:
I have tried some gif,  no one can be played with transparent background.  Do I miss something?
-Andy

crazytyt, 2021-01-22:
[image]

crazytyt, 2021-01-22:
This is what I got (1 frame)

-Andy

kisvegabor, 2021-01-23:
Thanks.
I've just fixed it. It works well for me with the "bird gif" you have sent.

crazytyt, 2021-01-25:
Thanks for your quick fix.
Somehow,  the fix seems not perfect. I have tried another gif and found the background is blink.
Please try this one at your side.
-Andy
[image]

kisvegabor, 2021-01-27:
I've checked it and the problem is that on the 6th frame the transparent backgrounds index is 60 while on the other frames it's 63. 

It messes up the LVGL images palette because the gif can change the palette for every frame but lvgl has a "global" palette for the whole image. See the [README of gifdec](https://github.com/lecram/gifdec) for more info.

Normally the rendered image should be stored as RGB888 or ARGB8888 but we store it as 8bit palette to save space but it doesn't work in some cases.

So we can either change to "correct" implementation with (A)RGB image output and accept the 5 times larger memory usage (24 or 32 bit output image + 2 internal 8 bit buffers) or accept the current limitations.

With the "correct method" a 100x100 gif would need 50kB RAM instead of 10 kB.

IMO we should choose the correct implementation instead.

What do you think?

crazytyt, 2021-03-04:
After I set gif->bgindex = 0xff,  which means the color index is 0xff which doesn't exist in most cases, the background is transparent.  Could you please explain it a little more?
Thanks
-Andy

kisvegabor, 2021-03-08:
It works because you are lucky. Theoretically, any color of the palette could be changed and the image could become a mess.

kisvegabor, 2021-04-14:
It should work in the new version.

closed 2021-04-14 by kisvegabor

jianjunxia, 2022-02-17:
> It works because you are lucky. Theoretically, any color of the palette could be changed and the image could become a mess

I seem to have a similar problem with you on the V7 version

commit 8d8359f1 (2021-01-23): fix(decoding): fix decoding gifs with transparent background

fixes #15