lvgl/lv_lib_gif #19: The output shows defects

guoweilkd, 2021-01-29:
Original gif
[image]

output gif
[image]

guoweilkd, 2021-01-31:
I have found the cause of the problem:when gif use Local Color Table and disposal==1,The gif will go wrong。In this case，Subsequent frames will change the color mapping of the previous frame in the gif lib.

kisvegabor, 2021-02-02:
Hi,

I added some optimizations to the original gif decoder to reduce RAM usage to ~1/6 but it turned out that it causes issues in some cases. See this comment: https://github.com/lvgl/lv_lib_gif/issues/15#issuecomment-768136218

I think the decoder should the reverted to the "original" way of working and we need to accept the large memory usage.

closed 2021-04-14 by guoweilkd