lvgl/lv_lib_gif #22: GIF with local color table may not display correctly?

crazytyt, 2021-03-04:
I have tried a GIF with local color table and found it can't be displayed correctly.
The program has just been  frozen. Only 1 or 2 frame was displayed on screen with distorted.
Here is my test gif.
Could you please try it on your side?
-Andy
[image]

kisvegabor, 2021-03-08:
I've reverted the original decoder in the `revert-original` branch but this gif crashed for me too. 
I've uploaded it to https://ezgif.com/ optimized and downloaded it and now works well.

With `revert-original` all the images (even [this](https://github.com/lvgl/lv_lib_gif/issues/15#issuecomment-766578609)) worked well for me.

`revert-original` supports only `LV_COLOR_DEPTH 32` and opening gif from data at this moment.

Can you test it? If it works for you too I'll finish it and merge to master.

crazytyt, 2021-03-09:
Unfortunately, it doesn't work on my side. Even the first frame of the GIF was not displayed and quickly returns "make: *** [run] Error -1073741819".
I've tested on LVGL simulator.

kisvegabor, 2021-03-09:
Have you reconverted the gif with ezgif as I suggested? Here is the gif that worked for me:

[image]

crazytyt, 2021-03-10:
Yes, if I re-converted this gif with ezgif, then it works well. 
But you said "With revert-original all the images (even this) worked well for me.". This isn't the case on my side.
I hope the original GIF can be decoded correctly.  It is not convenient if we "ezgif" it each time before GIF was used.
That must be somewhere in the decode source code that needs to be improved.

kisvegabor, 2021-03-10:
By "(even this)" I meant the kissing bears gif on the link. 

I don't know why this blue gif is not working. I'd required deep debugging of the gif decoder. I can't dive into it in the near future but if you have any ideas it would be welcome.

crazytyt, 2021-03-10:
Oops, sorry I didn't expect "THIS" is a link.

I need time to digest the code even though I want to fix this by myself.

kisvegabor, 2021-04-14:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2021-04-14 by kisvegabor