lvgl/lv_lib_gif #23: GIF cannot loop indefinitely

mlpzaq9998, 2021-04-15:
[image]

ui->Page_clock_img1 = lv_gif_create_from_file(ui->Page_clock, "S:/Astronaut.gif");

kisvegabor, 2021-04-15:
The gif library sends an `LV_EVENT_LEAVE` when the gif is finished. There you can call `lv_gif_restart(gif_obj)`

mlpzaq9998, 2021-04-15:
> The gif library sends an `LV_EVENT_LEAVE` when the gif is finished. There you can call `lv_gif_restart(gif_obj)`

Can the process be simplified? Parameterize or follow GIF settings

kisvegabor, 2021-04-19:
I've pushed some updates. Now it considers the loop count.

There some other changes too so some regression is possible. 
The previous version of gif decoder was buggy in some cases. Now the correct implementation is used but issues more memory.

closed 2021-07-08 by stale[bot]