lvgl/lv_lib_gif #24: No Data on the LCD

VanitarNordic, 2021-04-21:
Hi,

I used a library with this code:

```
extern const uint8_t mygif_map[];

.
.
.
.
![20210421_182148](https://user-images.githubusercontent.com/18719591/115565715-db3b6e80-a2ce-11eb-8d0a-94dc63402090.jpg)

lv_obj_t * par = lv_obj_create(lv_scr_act(), NULL); /*Create a parent object on the current screen*/
lv_obj_set_size(par, 320, 240);
lv_obj_t * img = lv_gif_create_from_data(par, mygif_map);

```

However, the LCD shows this:

kisvegabor, 2021-04-21:
Hi,

Please open an issue at https://github.com/lvgl/lv_lib_gif  and attach the
original gif too.

Thank you!

Regards,
Gabor

Hesam Moshiri ***@***.***> ezt írta (időpont: 2021. ápr. 21.,
Sze, 15:55):

VanitarNordic, 2021-04-21:
Hi,

Yes, I have opened the issue at the right address, which is here. I attached the original gif.

[image]

kisvegabor, 2021-04-21:
Uh sorry. Somehow I thought it's an email, not a GitHub notification... :sweat_smile: 

I'll test it tomorrow.

kisvegabor, 2021-04-22:
I've checked it and I see the gif but it freezes after ~10 frames. It seems it's a bug in the gif decoder.
I've opened an issue for it https://github.com/lecram/gifdec/issues/13

Does the decoder work with the example gif?
https://github.com/lvgl/lv_lib_gif/blob/master/example_gif.c

VanitarNordic, 2021-04-22:
I first tested this on the CodeBlock, as you said it freezes after the frames finish and when it tries to re-start the frames. I have not tested the example gif. I will do and inform you. 
For this experiment (my own gif), I don't know what hardware do you use, In my case, RAM and FLASH memories were not full, however, I received that message on the LCD. reason is unkown

VanitarNordic, 2021-04-22:
I tested the example GIF. it shows it on the screen, but no animation.

[image]

VanitarNordic, 2021-04-22:
This is memory consumption information for the example gif:

[image]

kisvegabor, 2021-04-26:
> I first tested this on the CodeBlock, as you said it freezes after the frames finish and when it tries to re-start the frames. I have not tested the example gif. I will do and inform you.

For me, it froze after 12 frames but the gif has > 20 frames. 

Please enable LVGL's logging in `lv_conf.h` with `LV_LOG_WARN` level and post the result.

VanitarNordic, 2021-04-26:
> For me, it froze after 12 frames but the gif has > 20 frames.
Please enable LVGL's logging in lv_conf.h with LV_LOG_WARN level and post the result.

Should I test with which gif file, the example gif, or my own gif?
on the hardware or on the CodeBlock?

kisvegabor, 2021-04-26:
With the example gif please, to find out why it doesn't run.

VanitarNordic, 2021-04-26:
Okay. I enabled the LVGL log like this with `printf`

[image]

I received these two warning in the console when the tracer:

```
Warn: Couldn't allocate memory 	(lv_mem.c #199 lv_mem_alloc())
Warn: Couldn't allocate memory 	(lv_mem.c #306 lv_mem_realloc())
```

VanitarNordic, 2021-04-26:
These are the memory configurations in `lv-conf.h`

`#define LV_DISP_ROT_MAX_BUF  (10U * 1024U)`
`#define LV_MEM_SIZE    (32U * 1024U)`

kisvegabor, 2021-04-26:
The gif decoding needs a lot of memory to store temporary data from the previous frames. I've pushed fix to slightly lower but with 16 bit color depth it's still  `4 x image_size` bytes. With the example gif, it's 4 x 60 x 80 = 16,000 plus some extra memory used during decoding the gif. 

Does it work if you increase `LV_MEM_SIZE` to `48U * 1024U`?

VanitarNordic, 2021-04-26:
Yes, it worked with 48U and I can see the animation. The speed of the animation is "a bit" slower than the actual test on a PC, however, the difference is not significant. The LCD bus is SPI which I flush the data using DMA. the main clock is 168MHz. 

is there any method to inform the LVGL engine to use external SRAM? I can connected an external SRAM to the MCU using FSMC and I know the SRAM start address would be something like 0xC00000000 for example.

kisvegabor, 2021-04-26:
> is there any method to inform the LVGL engine to use external SRAM? I can connected an external SRAM to the MCU using FSMC and I know the SRAM start address would be something like 0xC00000000 for example.

Yes, use `LV_MEM_ADR` in `lv_conf.h`. 
From the 0xC000000 address, I suppose you use an ST MCU. For someone, it wasn't working well. See [here](https://forum.lvgl.io/t/how-to-put-work-mem-in-sdrm/4146/21). But I assume it's "only" a memory configuration issue.

VanitarNordic, 2021-04-26:
> Yes, use LV_MEM_ADR in lv_conf.h.
From the 0xC000000 address, I suppose you use an ST MCU. For someone, it wasn't working well. See here. But I assume it's "only" a memory configuration issue.

Yes, I use STM32. Thank you for your support. 
By the way, this thread remains open till that freezing problem gets solved. The CodeBlock also freezes with that GIF file.

kisvegabor, 2021-04-30:
New comment here https://github.com/lecram/gifdec/issues/13

kisvegabor, 2021-05-04:
@lecram has fixed the issue of the gif decoder and I updated our decoder accordingly. 
Now the gif with the long-haired girl works well.

closed 2021-05-04 by kisvegabor