lvgl/lv_lib_gif #25: freeing previously played animations on screen

WiktorBAbove, 2021-06-18:
I have a intro screen that shows a pretty simple gif animation 320x240 5fps 4bit color.  about 48kb in size.
later on in my menu system i have another animation on a different screen that is similar but is 46kb.
The first animation is played ok, the second one just shows "No \ndata".
Upon trying to use a much smaller animation (the example_gif) the second animation is played good.
So my question is how do I in a good way delete the one-time-intro animation in a good way?
I've tried lv_obj_del, lv_img_cache_invalidate_src and even increasing the cache in lv_conf.h but it is always showing "No \ndata".
Also I am running on a STM32F429 discovery board.

Any wise advice || good tips I can try or am I using it incorrectly?

kisvegabor, 2021-06-20:
I've tested the latest version from `master` using lvgl v8 and it freed the memory correctly with a simple `lv_obj_del()`.

I suggest enabling `LV_USE_MEM_MONITOR` in `lv_conf.h` to see how the memory usage and fragmentation change.

WiktorBAbove, 2021-07-02:
Sorry for very late response,
I've tested` lvgl8` and it works quite well; took a moment to convert whole project to `lvgl8`.
Upon testing i noticed two things, animations that are `320x240, 8bit color, 12fps` arent working. probably due to the fact that its out of memory(works on `lvgl7`). it did however free correctly! which was the purpose of this issue.

kisvegabor, 2021-07-05:
Thanks for the feedback. 

The version of the gif decoder was incorrect To make it work correctly much more memory is required. :slightly_frowning_face: 

Anyway, as the original issue is solved I close it now.

closed 2021-07-05 by kisvegabor