lvgl/lv_lib_gif #29: Unknow references to LVGL

paloky, 2021-09-21:
Hi.
It's the firts time I use a LVGL, and I try to compile a simple example with the sample gift, but I have a compiler problems.
I use a ESP32, and I put the library inside the "components" folder of  "esp-idf".
I need to put the library on another path?

```
In file included from ../main/main.c:6:
../esp-idf/components/lv_lib_gif/lv_gif.h:33:5: error: unknown type name 'lv_img_t'
     lv_img_t img;
     ^~~~~~~~
../esp-idf/components/lv_lib_gif/lv_gif.h:35:5: error: unknown type name 'lv_timer_t'
     lv_timer_t * timer;
     ^~~~~~~~~~
../esp-idf/components/lv_lib_gif/lv_gif.h:40:14: error: unknown type name 'lv_obj_class_t'
 extern const lv_obj_class_t lv_gif_class;
              ^~~~~~~~~~~~~~
../main/main.c: In function 'main_task':
../main/main.c:75:13: warning: unused variable 'img' [-Wunused-variable]
  lv_obj_t * img = lv_gif_create_from_data(lv_scr_act(), example_gif_map);
```

kisvegabor, 2021-09-21:
Hi,
Be sure to use the release/v7 branch of this repo if you use LVGL v7.

paloky, 2021-09-21:
Hi kisvegabor.
I checkout to "release/v7" and now compile, but I get this error on a Linker.

`c:/users/paloky/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.main_task+0x10): undefined reference to `lv_gif_create_from_data'`

kisvegabor, 2021-09-21:
Maybe you need a `components.mk`? I don't know ESP in details, but it should be something related to it.

closed 2021-09-24 by paloky