lvgl/lv_lib_gif #30: lv_gif_event_cb callback don't work

paloky, 2021-09-24:
Hi.
I try to call a event after end the gif animation, but the callback don't work.
This is the code I use.
The GIF is showing corretly in the screen,
I use the "release/v7" branch.

What is wrong?

```
void TFT_ShowGIF()
{
	ESP_LOGD(TAG, "Showing GIF");
	
	extern const uint8_t img155_155_opt70_map[];  //Use the example gif
	lv_obj_t * gif1 = lv_gif_create_from_data(lv_scr_act(), img155_155_opt70_map);
	
	// Set Finish Event
	lv_obj_set_event_cb(gif1, lv_gif_event_cb);
	
	lv_obj_set_pos(gif1, 0, 0);         //Set its position	
}

void lv_gif_event_cb(struct _lv_obj_t * obj, lv_event_t event)
{
	ESP_LOGD(TAG, "GIF Event");
	
        if (LV_EVENT_LEAVE != event)  {
              return;
        }
  
       lv_obj_del(obj);
  }
```

kisvegabor, 2021-09-29:
The event should be sent [here](https://github.com/lvgl/lv_lib_gif/blob/release/v7/lv_gif.c#L126) but only if the gif is not looped infinitely.

paloky, 2021-09-29:
Hi kisvegabor.
I change the GIF for a only one loop using a EZGif.
In a PC, the GIF show only a one loop, but in the application, the GIF is showing in a infinite loop.

I need to config something for a one loop??

Thank's a lot.

kisvegabor, 2021-09-30:
There can be something problem [here](https://github.com/lvgl/lv_lib_gif/blob/release/v7/lv_gif.c#L110). Can you debug it?

closed 2022-04-28 by stale[bot]