lvgl/lv_lib_gif #37: i think next_frame_task_cb function has some problems in release/v7

jianjunxia, 2022-02-10:
in release/v7 next_frame_task_cb function，
i think this code has some problems:

Because I tried a lot of GIFs, its loop_count is zero, but this code does not send LV_EVENT_LEAVE signal when loop_count=0, so that the upper application can stop it, but keeps looping

https://github.com/lvgl/lv_lib_gif/blob/27a74507ee73ae462daf8f4d4c2f65b366fc9cfd/lv_gif.c#L120-L132

I think this code can be modified like this：
https://github.com/lvgl/lv_lib_gif/pull/36/files

closed 2022-04-28 by stale[bot]

glory-man, 2023-01-04:
> in release/v7 next_frame_task_cb function， i think this code has some problems:
>
> Because I tried a lot of GIFs, its loop_count is zero, but this code does not send LV_EVENT_LEAVE signal when [...]

loop_count = 0 means that animation should repeated infinitely. I think the problem is in current version
https://github.com/lvgl/lv_lib_gif/blob/afdc2d69c7e80210256fd84837b839320fe2618a/lv_gif.c#L120-L143
In my example animated gif with loop_count = 0 shows only 1 cycle after that `LV_SIGNAL_LEAVE` sended wich cause `lv_task_set_prio(ext->task, LV_TASK_PRIO_OFF);` and animation freezes.

kisvegabor, 2023-01-06:
It was just updated in LVGL. Probably it can be applied here too.
See https://github.com/lvgl/lvgl/issues/3883