lvgl/lv_lib_gif #42: The gif rendering takes too much memory

potato1992, 2022-12-14:
I see the code in `gd_render_frame(gifobj->gif, (uint8_t *)gifobj->imgdsc.data)`, a whole image is rendered at once and the color of every pixel is stored in `gifobj->imgdsc.data`.

I think a better way is to render a part of the image, say, 1/10 for example, then loop to render the whole image, this would save a lot of memory required to run this library. In most platforms for LVGL, the hardware RAM is very limited, and a 200x200 gif would eat up `LV_COLOR_DEPTH x 40000` bytes memory.

kisvegabor, 2022-12-15:
I agree, that it can be improved. 

My idea was to use the indexed image directly as LVGL can decode it out of the box. However the  palette can change on each frame and it is assumed that the pixels from the previous frame with the other palette are already written. 

If we added a limitation, i.e. "GIFs only with global palette can be used" the RAM usage would be `width x height` bytes. I'm sure there are tools to convert GIFs to global palette only format. 

I think it'd be the simplest.  What do you think?

potato1992, 2023-02-06:
> 

Sounds reasonable, it would surely bring a lot more chips to run lv_lib_gif if the RAM usage could be cut half!

remfan77, 2024-09-03:
Hello,
I have a similar problem.

**LVGL**
9.1

**PROBLEM**
At the boot of my embedded board I would like to show a splash gif 320x240.
The problem is that it require too much RAM, so it is not possible to attach a gif to a screen.

**ACTUAL SOLUTION**
Integrate a copy of gifdec and decode/draw the gif. But this is not a clean solution.

**QUESTION**
Is there a way to decode the image runtime without save into a buffer (using the gifdec library embedded in LVGL) ?
What is the cleanest solution using LVGL lib ?

Thanks for you great work.
Best regards,
Paolo

kisvegabor, 2024-09-04:
@bitbank2 has a decoder which is more optimal in therm of memory usage.

lanistor, 2025-03-11:
> [@bitbank2](https://github.com/bitbank2) has a decoder which is more optimal in therm of memory usage.

I saw https://github.com/bitbank2/AnimatedGIF, will it be embeded into LVGL?

lanistor, 2025-03-11:
And, why lv_lib_gif need `3 x image width x image height` memory for gif while `LV_COLOR_DEPTH=8`?
Why not just cache one frame of gif?

bitbank2, 2025-03-11:
I just wrote a very detailed example showing the tradeoffs between RAM usage and speed:
https://github.com/bitbank2/AnimatedGIF/tree/master/examples/perf_explained

For multi-frame GIFs with local palette changes, you'll need to hold a 'canvas' worth of pixels in the destination format (e.g. RGB565) and each new frame can add unique colors to any area, while the previous data needs to be preserved.

For GIF sequences with only a global palette, it's possible to only hold on to the 8-bit indexed pixels (still a canvas sized buffer), and then apply new pixels through the single global palette.

For situations where the display has it's own framebuffer (e.g. SPI LCD), you can get away with processing the GIF frames 1 line at a time and have the LCD hold onto the 'cooked' pixels for you.

Specifically for my AnimatedGIF library, it needs about 18K of RAM for the dictionary, file and pixel buffers. You could reduce that amount, but performance would suffer. Add on to that the memory needed to hold the 8-bit indexed data and/or the 'cooked' canvas pixels.

lanistor, 2025-03-12:
@bitbank2 Thanks a lot. Do you have a demo with `lvgl`?

I had a gif file sized `800*400`, with lv_lib_gif, it need nearly `1M` memory, we don't have that much memory left.

bitbank2, 2025-03-12:
> [@bitbank2](https://github.com/bitbank2) Thanks a lot. Do you have a demo with `lvgl`?
> 
> I had a gif file sized `800*400`, with lv_lib_gif, it need nearly `1M` memory, we don't have that much memory left.

I do not have a demo ready for LVGL. It would take some time to prepare even if the data can all fit in memory. It would be more efficient if you bypassed LVGL and just played GIF animation directly on the display.